LibreDWG/libredwg #1178: XRECORD object issue

michal-josef-spacek, 2025-09-02:
There are issues in XRECORD object with data types.

File is from test data defined in https://github.com/LibreDWG/libredwg/discussions/1028#discussioncomment-14207869
[Balken.dwg.gz](https://github.com/user-attachments/files/22098501/Balken.dwg.gz)

```
Next object: 238 Handleoff: 0x1 [UMC] Offset: 58131 [MC] @634638
==========================================
Object number: 238/EE, Size: 2047 [MS], Type: 79 [BS], Address: 634640
Add object XRECORD [238] Decode object XRECORD
bitsize: 16322 [RL] @3.2
 Hdlsize: 0x36, hdl_dat: @2040.2 - @2047.0 (2047)
handle: 0.2.2F3 [H 5]
EED[0] size: 0 (end)
num_eed: 0
num_reactors: 1 [BL 0]
ownerhandle: (12.1.15) abs:2DE [H 330]
reactors[0]: (4.2.2DE) abs:2DE [H* 330] => DICTIONARY
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 68
xdata_size: 2025 [BL 0]
  xdata[0] type: 90 [RS]
xdata[0]: 1359479961 [RLd 90]
  xdata[1] type: 90 [RS]
xdata[1]: 3615 [RLd 90]
  xdata[2] type: 90 [RS]
xdata[2]: 51 [RLd 90]
...
  xdata[261] type: 210 [RS]
xdata[261]: 0.660000 [RD 210]
  xdata[262] type: 4294959432 [RS]
ERROR: Invalid xdata type -7864 [RS]
cloning: 1 [BS 280]
num_objid_handles: 0 [BL]
 padding: +6
 object_map{2F3} = 238
 padding: 3F/3F (6 bits)
crc: 00EC [RSx]
 check_CRC 634638-636687 = 2049: 00EC == 00EC
```

The issue is with src/dwg.c dwg_resbuf_value_type() where we need return DWG_VT_POINT3D for group code 210 (instead of DWG_VT_REAL).
There are many issue with it 40, 140, 1040 etc.

michal-josef-spacek, 2025-09-02:
@rurban I implemented naive patch, but this is bad approach.
```
diff --git a/src/dwg.c b/src/dwg.c
index befd9cac0..a35188b34 100644
--- a/src/dwg.c
+++ b/src/dwg.c
@@ -1868,8 +1868,12 @@ dwg_resbuf_value_type (short gc)
                 return DWG_VT_BINARY;
               if (gc <= 1009)
                 return DWG_VT_STRING;
-              if (gc <= 1059)
+              if (gc <= 1010)
+                return DWG_VT_POINT3D;
+              if (gc <= 1042)
                 return DWG_VT_REAL;
+              if (gc <= 1059)
+                return DWG_VT_POINT3D;
               if (gc <= 1070)
                 return DWG_VT_INT16;
               if (gc == 1071)
@@ -1924,7 +1928,7 @@ dwg_resbuf_value_type (short gc)
       if (gc >= 210) // 210-299
         {
           if (gc <= 239)
-            return DWG_VT_REAL;
+            return DWG_VT_POINT3D;
           if (gc <= 269)
             return DWG_VT_INVALID;
           if (gc <= 279)
@@ -1940,6 +1944,10 @@ dwg_resbuf_value_type (short gc)
             return DWG_VT_HANDLE;
           if (gc <= 109)
             return DWG_VT_INVALID;
+          if (gc <= 142)
+            return DWG_VT_REAL;
+          if (gc <= 147)
+            return DWG_VT_POINT3D;
           if (gc <= 149)
             return DWG_VT_REAL;
           if (gc <= 169) // e.g. REQUIREDVERSIONS 160 r2013+
@@ -1954,8 +1962,10 @@ dwg_resbuf_value_type (short gc)
     {
       if (gc >= 38) // 38-102
         {
-          if (gc <= 59)
+          if (gc <= 42)
             return DWG_VT_REAL;
+          if (gc <= 59)
+            return DWG_VT_POINT3D;
           if (gc <= 79)
             return DWG_VT_INT16;
           if (gc <= 99)
```

rurban, 2025-09-02:
Well, do you want to implement a table of ranges and binary search into that? Good enough for now I think