LibreDWG/libredwg #1180: Heap buffer overflow in decompress_R2004_section

dkvhr, 2025-09-05:
Hello, I've found a heap-based buffer overflow crash through fuzzing. This happens with the normal configuration and compilation and also on a build with a release flag enabled.

This is the error:

```
Warning: checksum: 0xd0a61452 (calculated) page checksum mismatch 0x2dca0-0x2dcb4

Warning: Invalid section_array_size: [25].48 != 10556102
Warning: page_type:      0x7156b47b  (should be 0x4163043b)
Warning: page_type:      0x7154625b  (should be 0x4163043b)
Warning: summaryinfo_address mismatch: 808464432 != 288
Warning: page_type:      0x7156a8db  (should be 0x4163043b)
Warning: page_type:      0x7154e41b  (should be 0x4163043b)
Warning: page_type:      0x7154a39b  (should be 0x4163043b)
Warning: page_type:      0x7154b6bb  (should be 0x4163043b)
Warning: page_type:      0x7154b4bb  (should be 0x4163043b)
Warning: page_type:      0x71556ebb  (should be 0x4163043b)
[1]    119852 segmentation fault (core dumped)  ./programs/dwgread ../../libredwg/min1.dwg
```

This is the output with ASAN on a normal build:

```
=================================================================
==123541==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x53400001e8bb at pc 0x5c0a7a02aa8e bp 0x7ffc6ff22180 sp 0x7ffc6ff22178
WRITE of size 1 at 0x53400001e8bb thread T0
    #0 0x5c0a7a02aa8d in decompress_R2004_section /libredwg/src/decode.c:1313:27
    #1 0x5c0a7a02b9b7 in read_2004_compressed_section /libredwg/src/decode.c:2167:19
    #2 0x5c0a7a00bbec in read_2004_section_handles /libredwg/src/decode.c:2457:11
    #3 0x5c0a7a00bbec in decode_R2004 /libredwg/src/decode.c:3630:12
    #4 0x5c0a79fc982f in dwg_decode /libredwg/src/decode.c:231:12
    #5 0x5c0a79f80c25 in dwg_read_file /libredwg/src/dwg.c:277:11
    #6 0x5c0a79f7f5d9 in main /libredwg/programs/dwgread.c:260:15
    #7 0x7b3faf82a1c9 in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #8 0x7b3faf82a28a in __libc_start_main csu/../csu/libc-start.c:360:3
    #9 0x5c0a79e99d44 in _start (/libredwg/programs/dwgread+0x408d44) (BuildId: ae3d9222fb14482f520d48742868ca58cc142616)

0x53400001e8bb is located 0 bytes after 123067-byte region [0x534000000800,0x53400001e8bb)
allocated by thread T0 here:
    #0 0x5c0a79f39f90 in realloc (/libredwg/programs/dwgread+0x4a8f90) (BuildId: ae3d9222fb14482f520d48742868ca58cc142616)
    #1 0x5c0a79f9d5a7 in bit_chain_alloc_size /libredwg/src/bits.c:3957:30
    #2 0x5c0a79f9d5a7 in bit_chain_alloc /libredwg/src/bits.c:3979:3
    #3 0x5c0a79f9d5a7 in bit_write_RC /libredwg/src/bits.c:311:9

SUMMARY: AddressSanitizer: heap-buffer-overflow /libredwg/src/decode.c:1313:27 in decompress_R2004_section
Shadow bytes around the buggy address:
  0x53400001e600: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x53400001e680: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x53400001e700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x53400001e780: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x53400001e800: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x53400001e880: 00 00 00 00 00 00 00[03]fa fa fa fa fa fa fa fa
  0x53400001e900: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x53400001e980: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x53400001ea00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x53400001ea80: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x53400001eb00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==123541==ABORTING
```

To reproduce, you can run `./programs/dwgread min.dwg`
Where the crash file can be found here:
[crash.tar.gz](https://github.com/user-attachments/files/22177628/crash.tar.gz)

Please feel free to reach out if you have any inquiries about this!

SHAIKH-AZAZ, 2025-09-05:
How to setup this?

dkvhr, 2025-09-05:
> How to setup this?

You can setup it with:

```
./autogen.sh
./configure --enable-release
make -C src
make -C programs dwgread
```

SHAIKH-AZAZ, 2025-09-05:
Editor to use ? 
And which os to prefer? 
Any suggestions

rurban, 2025-09-06:
Yeah, the decompressor is completely new, so fuzzing will find new overflows. Thanks