LibreDWG/libredwg #1183: Buffer overflow in json_acis_data detected on Fedora

neilberkman, 2025-09-05:
Buffer overflow detected when processing JSON files with ACIS data arrays.

## Error
```
Reading JSON file example_r14.json
*** buffer overflow detected ***: terminated
```

## Location
`src/in_json.c` function `json_acis_data` - incorrect buffer allocation calculation.

## Details
The function allocates based on JSON array token size (`t->end - t->start`) rather than the actual combined string content size, causing overflow when concatenating strings.

## Reproduction
Build on Fedora with default flags (includes `-D_FORTIFY_SOURCE=2`) and run:
```bash
dwgread -o example_r14.json test/test-data/example_r14.dwg
dwgwrite -o test.dwg example_r14.json
```

## Evidence
CI failure: https://github.com/neilberkman/libredwg/actions/runs/17504170044/job/49724000228#step:10:170

Discovered while implementing multi-platform CI (#1181).

rurban, 2025-09-06:
I am on Fedora, but cannot repro yet. The description sounds also suspicious

rurban, 2025-09-07:
Yes, strange, Investigating a compiler bug (wrong overflow error)

neilberkman, 2025-09-07:
> Yes, strange, Investigating a compiler bug (wrong overflow error)

I've seen this overflow occur specifically when `FORTIFY_SOURCE=2` is enabled (Fedora's default). This flag adds runtime bounds checking that may be catching a real overflow that other systems miss.

You could try compiling with `-U_FORTIFY_SOURCE` or `-D_FORTIFY_SOURCE=0` to see if the issue disappears. If it does, it suggests the overflow exists but is only detected with `FORTIFY_SOURCE` enabled.

The allocation calculation in `json_acis_data` appears to use the JSON token size rather than summing the actual string lengths needed for concatenation, which could be the root cause.